AGHSEagleRobotics/frc1388-2023 #7: Use Lint to test balancing

jtechau, 2023-01-12:
- [ ] clone lint repository
- [ ] write test balancing code
- [ ] test

jtechau, 2023-01-16:
Let's keep track of ideas and test results / notes in the description above.

isabellebowdey, 2023-01-24:
We coded AutoBalance on the Lint2023 project to drive up if the robot is tilted up (to get up the charging station). 
We used power-based code, multiplying a P value (0.05 starting) by the robot angle. The steeper the angle, the faster it goes. This could create some overshoot problems, so we're thinking of ways to balance on the charge station more successfully. Some ideas (a few mentioned above):
- use the power-based ( P times angle ) code, but stop the robot when it gets to a certain angle / value
- use a constant speed regardless of angle, and stop once the angle hits close to 0 or the encoder hits a certain distance traveled
- - use a full PID controller for more accuracy, use the D (derivative ) value specifically

jtechau, 2023-03-08:
Autobalance has been ported to the 2023 robot code.