AGHSEagleRobotics/frc1388-2023 #8: Dash Board

calvinxp, 2023-01-14:
- figure out what has to be on the dashboard
- [ ] auto selector
- [ ] camera stream
- [ ] gyro angle for balancing

calvinxp, 2023-01-18:
Currently, the secondary camera on USB port 1 won't stream to shuffleboard, but the primary camera still works.

jtechau, 2023-02-15:
Note the following from https://docs.wpilib.org/en/latest/docs/yearly-overview/yearly-changelog.html#breaking-changes:

 - NetworkTables 4.0 (NT4) introduced several breaking changes. Shuffleboard classes now return GenericEntry instead of NetworkTableEntry; as GenericEntry provides nearly all the same methods, a simple textual replacement of the class name should suffice.