AGHSEagleRobotics/frc1388-2023 #24: IMU class should use the robot's orientation

jtechau, 2023-01-29:
The IMU classes (e.g. `IMUSubsystem16470`, `IMUSubsystem16448`) include methods to return values for the three gyro axes.  These methods should represent rotation and direction relative to the robot's coordinate system rather than the sensor's coordinate system.

This means, for example, that the IMU class should translate the axes as appropriate so that the `getYGyro()` method returns the angle of the robot's left-right axis, where tipping the front of the robot up is a negative rotation.  Likewise, the `getZGyro()` method should return the gyro's heading, where a robot counter-clockwise rotation (looking down from above) is positive.  These translations may/will depend on the type of IMU used and it's orientation within the robot.

The orientation (and possibly type) of the IMU in this year's robot isn't yet known.  This issue should only be closed after the orientation of the IMU in the robot has been assigned and the IMU class updated accordingly.

As a placeholder, since we're currently using Lint for testing, the translation should be defined to match the type/orientation of the IMU on Lint.

https://docs.wpilib.org/en/stable/docs/software/advanced-controls/geometry/coordinate-systems.html#robot-coordinate-system
and
https://docs.wpilib.org/en/stable/docs/yearly-overview/yearly-changelog.html#breaking-changes (see Danger box)

MasonGomez1388, 2023-02-01:
IMU16448's X and Z axis are opposite of IMU16470's

jtechau, 2023-02-01:
See also:  https://www.chiefdelphi.com/t/analog-devices-adis-16470-eng-basic-java-gyro-code-help/422553